autonomousvision/kitti360Scripts #129: Missing 3d Bouding box?

xixinzhang, 2024-08-30:
I have some questions about the labels of 3d BBox:

1. I found the start frame(2897) of boxes in sequences **4** does not match the start frame(0) of the LiDAR scan. Is this something expected?
2. I've also seen a similar issue https://github.com/autonomousvision/kitti360Scripts/issues/40, which is related to data of sequence 0. There seem to be a lot of missing bboxes, I haven't checked all frames of all sequences, but here are some examples below. I don't know if I missed some data or if they were just not given, but the case I give for sequence 4 is out of the range of the first 250 frames.

- sequence 2 (frame 4433 front view):
  [image]

- sequence 4 (frame 2907 front view)
  [image]
  (frame 3368 front view, no label at all)
  [image]
  (frame 3368 back view, no label at all)
  [image]
  Only three static cars are being labeled for the frame window [3356, 3586], which is wrong from the above visualization
  ```xml
  <object1100>
    <index>8576</index>
    <label>car</label>
    <semanticId_orig>7</semanticId_orig>
    <semanticId>13</semanticId>
    <instanceId>129</instanceId>
    <category>instance</category>
    <timestamp>-1</timestamp>
    <dynamic>0</dynamic>
    <transform type_id="opencv-matrix">
      <rows>4</rows>
      <cols>4</cols>
      <dt>f</dt>
      <data>
        1.14138556e+00 2.14470959e+00 -6.63366588e-03 3.82886426e+03
        -5.34604931e+00 4.57898617e-01 -1.12612871e-03 -1.66201196e+03
        9.46977234e-04 9.00010951e-03 1.63808441e+00 2.39120895e+02 0. 0.
        0. 1.</data></transform>
    <vertices type_id="opencv-matrix">
      <rows>8</rows>
      <cols>3</cols>
      <dt>f</dt>
      <data>
        5.00000000e-01 5.00000000e-01 5.00000000e-01 5.00000000e-01
        5.00000000e-01 -5.00000000e-01 5.00000000e-01 -5.00000000e-01
        5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        -5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        -5.00000000e-01 -5.00000000e-01 -5.00000000e-01 5.00000000e-01</data></vertices>
    <faces type_id="opencv-matrix">
      <rows>12</rows>
      <cols>3</cols>
      <dt>u</dt>
      <data>
        0 2 1 2 3 1 4 6 5 6 7 5 4 5 1 5 0 1 7 6 2 6 3 2 5 7 0 7 2 0 1 3 4
        3 6 4</data></faces>
    <start_frame>3356</start_frame>
    <end_frame>3586</end_frame></object1100>
  <object1101>
    <index>8577</index>
    <label>car</label>
    <semanticId_orig>7</semanticId_orig>
    <semanticId>13</semanticId>
    <instanceId>130</instanceId>
    <category>instance</category>
    <timestamp>-1</timestamp>
    <dynamic>0</dynamic>
    <transform type_id="opencv-matrix">
      <rows>4</rows>
      <cols>4</cols>
      <dt>f</dt>
      <data>
        4.78651315e-01 1.92826843e+00 7.02159759e-03 3.82994263e+03
        -4.01186228e+00 2.29451627e-01 -1.06386833e-01 -1.66836865e+03
        -3.10723543e-01 7.85248913e-03 1.38441491e+00 2.38654770e+02 0. 0.
        0. 1.</data></transform>
    <vertices type_id="opencv-matrix">
      <rows>8</rows>
      <cols>3</cols>
      <dt>f</dt>
      <data>
        5.00000000e-01 5.00000000e-01 5.00000000e-01 5.00000000e-01
        5.00000000e-01 -5.00000000e-01 5.00000000e-01 -5.00000000e-01
        5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        -5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        -5.00000000e-01 -5.00000000e-01 -5.00000000e-01 5.00000000e-01</data></vertices>
    <faces type_id="opencv-matrix">
      <rows>12</rows>
      <cols>3</cols>
      <dt>u</dt>
      <data>
        0 2 1 2 3 1 4 6 5 6 7 5 4 5 1 5 0 1 7 6 2 6 3 2 5 7 0 7 2 0 1 3 4
        3 6 4</data></faces>
    <start_frame>3356</start_frame>
    <end_frame>3586</end_frame></object1101>
  <object1103>
    <index>8579</index>
    <label>car</label>
    <semanticId_orig>7</semanticId_orig>
    <semanticId>13</semanticId>
    <instanceId>131</instanceId>
    <category>instance</category>
    <timestamp>-1</timestamp>
    <dynamic>0</dynamic>
    <transform type_id="opencv-matrix">
      <rows>4</rows>
      <cols>4</cols>
      <dt>f</dt>
      <data>
        -1.57342708e+00 -1.90071630e+00 -6.44791964e-03 3.83258594e+03
        4.33695078e+00 -6.89578295e-01 -1.15723989e-03 -1.65412012e+03
        -3.29058268e-03 -8.37895181e-03 1.55791259e+00 2.39629547e+02 0.
        0. 0. 1.</data></transform>
    <vertices type_id="opencv-matrix">
      <rows>8</rows>
      <cols>3</cols>
      <dt>f</dt>
      <data>
        5.00000000e-01 5.00000000e-01 5.00000000e-01 5.00000000e-01
        5.00000000e-01 -5.00000000e-01 5.00000000e-01 -5.00000000e-01
        5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        -5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        5.00000000e-01 5.00000000e-01 -5.00000000e-01 -5.00000000e-01
        -5.00000000e-01 -5.00000000e-01 -5.00000000e-01 5.00000000e-01</data></vertices>
    <faces type_id="opencv-matrix">
      <rows>12</rows>
      <cols>3</cols>
      <dt>u</dt>
      <data>
        0 2 1 2 3 1 4 6 5 6 7 5 4 5 1 5 0 1 7 6 2 6 3 2 5 7 0 7 2 0 1 3 4
        3 6 4</data></faces>
    <start_frame>3356</start_frame>
    <end_frame>3586</end_frame></object1103>
  ```

yiyiliao, 2024-10-07:
Thank you for reporting the issues! 

> I found the start frame(2897) of boxes in sequences 4 does not match the start frame(0) of the LiDAR scan. Is this something expected?

Yes, it is expected. We have withheld some RGB images from sequence 4 because they are used as test sequences for the NVS benchmark.

> I've also seen a similar issue https://github.com/autonomousvision/kitti360Scripts/issues/40, which is related to data of sequence 0. There seem to be a lot of missing bboxes, I haven't checked all [...]

I investigated this issue and found that the missing bounding boxes are likely due to retrieving them using the annotation `window`. Some regions have multiple annotations because the ego-car passed the same areas repeatedly or because of overlaps between adjacent windows. To address this, we implemented an automatic check to filter out duplicate annotations, which might explain why some boxes appear from a different `window`. Below are screenshots showing the same sequence overlaid with instance bounding boxes, comparing retrieval using the annotation window versus retrieval based on distance."

[image]
1) Annotation `window`-based retrieval

[image]
2) Distance-based retrieval

yiyiliao, 2024-10-07:
I have updated `kitti360Viewer3D.py` to visualize the bounding boxes based on distance instead of annotation window:
https://github.com/autonomousvision/kitti360Scripts/blob/d4d4885102f72ea8d96c9e72e2ff03a8834353a4/kitti360scripts/viewer/kitti360Viewer3D.py#L319-L345

Liu2022, 2025-01-02:
So you mean there is something wrong with the 3Dbox XML file?

> Thank you for reporting the issues!
> 
> > I found the start frame(2897) of boxes in sequences 4 does not match the start frame(0) of the LiDAR scan. Is this something expected?
> 
> Yes, it is expected. We have withheld some RGB images from sequence 4 because they are used as test sequences for the NVS benchmark.
> 
> > I've also seen a similar issue #40, which is related to data of sequence 0. There seem to be a lot of missing bboxes, I haven't checked all frames of all sequences, but here are some examples below. I don't know if I missed some data or if they were just not given, but the case I give for sequence 4 is out of the range of the first 250 frames.
> 
> I investigated this issue and found that the missing bounding boxes are likely due to retrieving them using the annotation `window`. Some regions have multiple annotations because the ego-car passed the same areas repeatedly or because of overlaps between adjacent windows. To address this, we implemented an automatic check to filter out duplicate annotations, which might explain why some boxes appear from a different `window`. Below are screenshots showing the same sequence (seq 4, 0000003356_0000003586.ply) overlaid with instance bounding boxes, comparing retrieval using the annotation window versus retrieval based on distance."
> 
> ![ScreenCapture_2024-10-07-15-31-26](https://private-user-images.githubusercontent.com/15853014/374083289-f21980db-6083-42cc-8854-a990b0b37a8f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3ODM4MzEsIm5iZiI6MTczNTc4MzUzMSwicGF0aCI6Ii8xNTg1MzAxNC8zNzQwODMyODktZjIxOTgwZGItNjA4My00MmNjLTg4NTQtYTk5MGIwYjM3YThmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAyVDAyMDUzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmZmE5ZWJlNGFiYmU3MDRkMmUxNDk2OTliMGQ1ZDhmODdjZWE5ZjYwMzgyN2ZjOTExNzI4NDFkYjg0ZWQ4NjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.EaTcR8UFgAKB76zQLOxIF80QnPZrR8Jo1j5ouVHZZMg)
> 
> 1. Annotation `window`-based retrieval
> 
> ![ScreenCapture_2024-10-07-15-27-11](https://private-user-images.githubusercontent.com/15853014/374082920-6b47f39e-c2e9-4ec1-ab2c-1ec53cfecb78.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzU3ODM4MzEsIm5iZiI6MTczNTc4MzUzMSwicGF0aCI6Ii8xNTg1MzAxNC8zNzQwODI5MjAtNmI0N2YzOWUtYzJlOS00ZWMxLWFiMmMtMWVjNTNjZmVjYjc4LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTAyVDAyMDUzMVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTU0NzdmNWJiY2FkMjIzMmZkODU5N2NhN2ZlZjRkYjdkYTU4YTQ5YzEyMGI1NDFmMWE3ZThkYzhlZWFjNjEyMTMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.9sdYh2ETfQ_wUFTUMxNh_OpQewBQPYzhVkYefQZTvH8) 2. Distance-based retrieval

So you mean there is something wrong with the 3Dbox XML file?