autonomousvision/kitti360Scripts #131: Request of Ground Truth data for Geometric and Semantic Mapping val and train split

QingCheng24, 2024-10-07:
Hi, thank you for the great benchmark!

we want to evaluate our method on the [Geometric and Semantic Mapping](https://www.cvlibs.net/datasets/kitti-360/leaderboard_semantic_slam.php?task=mapping) task. But there's no Ground Truth data for `ground_truth_observed_region` and `ground_truth_conf` used in this [evaluation script].(https://github.com/autonomousvision/kitti360Scripts/blob/master/kitti360scripts/evaluation/semantic_slam/evalSemanticSLAM.sh)

Thus we ask for these two types of data in order to perform more local evaluation instead of using the evaluation server which provides very limited times per month. 

It is also great, if you can provide the processing script!

Much appreciated!

QingCheng24, 2024-10-11:
Hi Prof. @yiyiliao, could you please give some hints on how to reconstruct the GT data and run the script? 

I've built the binary and run it with the same kind of file for both GT and prediction. The files are created based on `data_3d_semantics` data. But the semantic evaluation failed with a confusion matrix full of zero.

yiyiliao, 2024-10-12:
Hi, thank you for evaluating your method on our benchmark. We withheld the ground truth data for fair comparisons. Could you send me a data example of your result via email (yiyi.liao@zju.edu.cn)? I will look into this.