autonomousvision/kitti360Scripts #132: Is there a small delay in the synchronization between the radar and camera？

tianjun-world, 2024-10-08:
Thank you very much for your work.
 During the visualization process, it was found that some 3D frames were misaligned with the actual position of the image. Compared to the actual position of the image, the 3D frames were positioned backwards. Is there a slight delay in the synchronization between radar and camera。

tianjun-world, 2024-10-08:
[image]

yiyiliao, 2024-10-08:
Thank you for your interest! The synchronization should be fairly accurate. Would you mind uploading the image again? It is not shown correctly on my side:
[image]

tianjun-world, 2024-10-08:
the 3D Box were positioned backwards
[image]

tianjun-world, 2024-10-08:
[image]