autonomousvision/kitti360Scripts #136: how to just download one sequence

pwjoin, 2025-04-25:
Thank you for your great work, the total data is too big, i just want download part of them,or just want dowanload one sequence.

yiyiliao, 2025-04-27:
Thank you for your interest. For the large files (e.g., `Perspective Images for Train & Val (128G)` and `Raw Velodyne Scans (119G)`), we provide a download script that fetches them sequence by sequence. If you'd like to download only specific sequences, you can simply comment out the ones you don't need.

yiyiliao, 2025-08-04:
Hi @kailqq, you may click on this to get the download script:

[image]